SBEMimage/SBEMimage #8: Stage position indicator does not work correctly when using Merlin stage

btitze, 2019-02-21:


templiert, 2019-08-31:
- The Get Stage X in the test tab is working correctly: the value matches the smartsem value. The viewport values also match smartsem stage values.

- The red cursor is not updated during manual stage move, also not when making actions in the viewport (which should involve repaints). It seems to be updated only after a stage move after right click in the viewport to move to a certain position. In this case the updating comes too late as shown in the scenario below:

-- right click in the viewport to move to 60k,65k : the stage moves correctly but the red cursor is updated after the move and placed in 60k,62k (it was the previous stage position)
-- right click in the viewport to move to 50k,50k: the stage  moves correctly but the red cursor is updated after the move and placed in 60k,65k (previous position)
-- right click in the viewport to move to 60k,65k: the stage  moves correctly but the red cursor is updated after the move and placed in 50k,50k (previous position)

btitze, 2019-11-11:
This should work now ([see this commit](https://github.com/SBEMimage/SBEMimage/commit/3312f227a5ff11eff6ce00ecb10442f21b4d9e44); the new coordinates were not read into self.last_known_x and self.last_known_y...)
Perhaps you could test it on your stage?