SBEMimage/SBEMimage #10: Issue with setting grid_size for new grids

templiert, 2019-07-02:
To create new grids of a certain size ([3,2] here) I do :

```
# remove existing grids except first one
self.gm.delete_all_additional_grids()
for section in range(n_sections - 1):
	self.gm.add_new_grid()
# set grid location, grid rotation, and grid size 
for idx, section in sections.items():
	self.cs.set_grid_origin_s(grid_number=idx, s_coordinates=list(map(float, section['center'])))
	self.gm.set_rotation(grid_number=idx, rotation=float(section['angle']))
	self.gm.set_grid_size(grid_number=idx, size=[3,2])
```
In the viewport, I get grids with a [5,1] shape instead of [3,2].

I checked add_new_grid (which by default first sets a [5,5] size) and update_active_tiles_to_new_grid_size but I cannot find what is wrong. Any idea ?

btitze, 2019-07-03:
Hi Thomas,

you need to call a few additional methods:
After changing the grid size, the grid map must be recalculated:
```
self.gm.set_grid_size(grid_number, ...)
self.gm.calculate_grid_map(grid_number)

```
And whenever you make changes to the grids, you need to update the viewport as follows to make these changes visible:
```
self.viewport.update_grids()
self.viewport.mv_draw()

```
If you add these methods, it should work (I just tested it).
For adding a (very) large number of grids, another approach may be better, though... let's perhaps talk on the phone tomorrow or on Friday?

templiert, 2019-07-03:
Perfect, it's working, thanks.

I'm currently playing with 500 sections: so far importing looks fast and viewport browsing looks seamless.

Sure, I'm free, call whenever it suits you.

closed 2019-07-03 by btitze