SBEMimage/SBEMimage #11: anchoring top-left of imported image to (0,0)

templiert, 2019-07-05:
(use_microtome = False)

To anchor an imported image to (0,0) I do [(here in context)](https://github.com/SBEMimage/SBEMimage/blob/magc-integration/src/dlg_windows.py#L879-L892): 

```
new_img_number = self.ovm.get_number_imported()
self.ovm.add_imported_img()
width, height = imported_img.size
self.cs.set_imported_img_centre_s(
	new_img_number,
	[width//2, height//2])
self.ovm.set_imported_img_file(
	new_img_number, target_path)
self.ovm.set_imported_img_name(new_img_number, self.lineEdit_name.text())
self.ovm.set_imported_img_size_px_py(
	new_img_number, width, height)
self.ovm.set_imported_img_pixel_size(
	new_img_number, 1000)
```
but there seems to be a small offset as seen on the picture below. Am I missing something ?

[image]

btitze, 2019-07-08:
I think this may have to do with the stage calibration. If the imported image is very large, even small discrepancies would explain such an offset. 
Perhaps it is best to anchor the wafer image at the top left corner? (The reason I chose the centre of the imported images for anchoring was because of the rotations.)

templiert, 2019-07-08:
> I think this may have to do with the stage calibration.

It could be yes.

I am looking at how you transform coordinates between the 

- stage 
- device
- viewport

coordinate systems and I am confused by a few things, I'm opening another issue.

> If the imported image is very large, even small discrepancies would explain such an offset.

Larger images indeed give larger offsets.

> (The reason I chose the centre of the imported images for anchoring was because of the rotations.)

I support this choice: dealing with centers is easier when rotations are involved.

templiert, 2019-08-12:
> I think this may have to do with the stage calibration.

For the record, with 
```
stage_scale_factor_x = 1
stage_scale_factor_y = 1
stage_rotation_angle_x = 0.000000001
stage_rotation_angle_y = 0.000000001
```
there is still an offset :

[image]

templiert, 2019-08-12:
Sorry, disregard my previous comment, stupid mistake on my side: the imported image itself has an offset.

So with 

```
stage_scale_factor_x = 1
stage_scale_factor_y = 1
stage_rotation_angle_x = 0.000000001
stage_rotation_angle_y = 0.000000001
```
[image]

there is no visible offset. Sorry for the noise.

closed 2019-08-13 by btitze