SBEMimage/SBEMimage #12: overlay grid in front of imported image?

templiert, 2019-07-05:
Regardless of the order of the sequence of adding a grid and importing an image, the imported image ends up in front of the grid.  Can the grid be brought in front of the imported image (without playing with the image transparency)?

btitze, 2019-07-08:
We would need to introduce a new option in the settings for that. Then it could be done easily. When the viewport is redrawn, the sequence of placing grids and overview images would be reversed. If more fine-grained control is required we could introduce a specified order for the different layers/objects, but this is probably overkill for now.

I think the easiest fix for your use case is to check in the method viewport.mv_draw() whether the MagC mode is active, and depending on that change the order of placing the imported image(s).

templiert, 2019-07-08:
Thanks, [changed here](https://github.com/SBEMimage/SBEMimage/commit/1999b0d708a7e5f14dac0cb9160ce37798bd2aff).