SBEMimage/SBEMimage #13: design: extending the config file?

templiert, 2019-07-07:
I find the config file quite practical to 
- store variables across sessions
- access variables across different modules in the code (e.g. knowing from inside the viewport whether some items are selected in the main_window)

but the config layout [seems to be very strict](https://github.com/SBEMimage/SBEMimage/blob/master/src/config_template.py), specifying the exact number of sections and keys.

I am thinking of modifying config_template.py to check whether there is a magc flag in config (e.g. magc_mode = True), then adjusting the required config layout. It would leave the standard sbemimage part unchanged. Would that be ok, or should I look for another design (e.g. a separate magc config file) ? 
Thanks

btitze, 2019-07-08:
I haven't made up my mind about this. 
For now, you could do this: 
Add a new section [magc] at the end of the default.ini config file and add the desired keys you need for MagC with default values. Also, add magc_mode = True in the section [sys] at the top.
Then increase the number of sections and keys accordingly in config_template.py
Old user configuration files will now be updated automatically when you start SBEMimage.

templiert, 2019-07-08:
Ok thanks, I have now a magc_mode key in [sys] and another section [magc].
I updated config_template [like this](https://github.com/SBEMimage/SBEMimage/blob/9c8eb5369184fe6724359815b657e114286a25b3/src/config_template.py#L19-L27):

```
CFG_NUMBER_SECTIONS_CORE = 10
CFG_NUMBER_KEYS_CORE = 194

CFG_NUMBER_SECTIONS_MAGC = 1 # magc
CFG_NUMBER_KEYS_MAGC = 4 # [sys][magc_mode],[magc][selected_sections] 
                        # [magc][checked_sections], [magc][sections]

CFG_NUMBER_SECTIONS = CFG_NUMBER_SECTIONS_CORE + CFG_NUMBER_SECTIONS_MAGC
CFG_NUMBER_KEYS = CFG_NUMBER_KEYS_CORE + CFG_NUMBER_KEYS_MAGC
```

closed 2019-08-15 by btitze