SBEMimage/SBEMimage #14: Transformations between {stage, device, viewport} coordinate systems ?

templiert, 2019-07-08:
I understand that there are 3 coordinate systems: 
- stage
- device (microtome or sem)
- viewport

Is it true that there should be an affine transform mapping for each pair of coordinate systems (except if there is some mirroring, for example a flipped x-axis, in which case there is a flip convolved with an affine transform) ?

If yes, then I fail to recognize affine transform machinery in the [variables stage_rotation_angle  and A,B,C,D](https://github.com/SBEMimage/SBEMimage/blob/master/src/coordinate_system.py#L61-L69), what are these ?
```
rot_x = float(self.cfg[device]['stage_rotation_angle_x'])
rot_y = float(self.cfg[device]['stage_rotation_angle_y'])
self.scale_x = float(self.cfg[device]['stage_scale_factor_x'])
self.scale_y = float(self.cfg[device]['stage_scale_factor_y'])
angle_diff = rot_x - rot_y
self.A = cos(rot_y) / cos(angle_diff)
self.B = sin(rot_x) / cos(angle_diff)
self.C = (-1) * sin(rot_y) / cos(angle_diff)
self.D = cos(rot_x) / cos(angle_diff)
```
Do these variables correspond to standard affine transform parameters, for example as used in the [java convention for affine transforms](https://docs.oracle.com/javase/7/docs/api/java/awt/geom/AffineTransform.html) with 
- scale (m00, m11)
- shear (m01, m10)
- translate (m02, m12)
factors ?
[image]

Also, one suspicious thing: using the following stage configuration
```
stage_scale_factor_x = 1
stage_scale_factor_y = 1
stage_rotation_angle_x = 0
stage_rotation_angle_y = 0
```
leads to a division by 0 in [convert_to_d](https://github.com/SBEMimage/SBEMimage/blob/master/src/coordinate_system.py#L89-L97) when calculating dx:
```
    def convert_to_d(self, s_coordinates):
        """Convert stage coordinates into SEM coordinates"""
        stage_x, stage_y = s_coordinates
        stage_x /= self.scale_x
        stage_y /= self.scale_y
        dx = ((stage_y + stage_x * (-self.D/self.C))/
              (self.B + self.A * (-self.D/self.C)))
        dy = ((stage_y - dx * self.B) / self.D)
        return (dx, dy)
```

btitze, 2019-08-08:
Yes, there are three coordinate systems: 
SEM coordinates in microns,
stage coordinates in microns,
and viewport coordinates (in pixels, for the current viewing window).
Two transformations are taken into account: 
- Scaling (scale factors for the x and the y axis)
- Rotation (angles for the x and the y axis).
I've derived the coordinate transformations myself, so it's a custom solution. I apply the scale factors to the axes and use a rotation matrix for the angles.
I'll write up an explanation with an illustration for the next version of the manual.
Thanks for pointing out the division by zero! The function convert_to_d() is the inverse of convert_to_s(), in which I perform a matrix multiplication. If there are no rotations, only the scale factors are needed. I'll look into fixing that.

templiert, 2019-08-12:
> If there are no rotations, only the scale factors are needed. I'll look into fixing that.

One suggestion:

[here I have 3 simple functions](https://github.com/SBEMimage/SBEMimage/blob/magc-integration/src/utils.py#L303-L317) to handle affine transforms which are robust to special cases (no rotation, etc.) :
```
def affineT(x_in, y_in, x_out, y_out):
    X = np.array([[x, y, 1] for (x,y) in zip(x_in, y_in)])
    Y = np.array([[x, y, 1] for (x,y) in zip(x_out, y_out)])
    aff, res, rank, s = np.linalg.lstsq(X, Y)
    return aff

def applyAffineT(x_in, y_in, aff):
    input = np.array([ [x, y, 1] for (x,y) in zip(x_in, y_in)])
    output = np.dot(input, aff)
    x_out, y_out = output.T[0:2]
    return x_out, y_out
    
def invertAffineT(aff):
    return np.linalg.inv(aff)
```
- compute a transform based on corresponding points (at least 3)
- apply a transform to points
- invert a transform

Calibrating the system would mean to compute
- transform_sem_to_stage
- transform_sem_to_viewport

with sets of matching points.

The [current conversion functions](https://github.com/SBEMimage/SBEMimage/blob/magc-integration/src/coordinate_system.py#L82-L103)
```
def convert_to_s(self, d_coordinates):
	"""Convert SEM coordinates into stage coordinates"""
	dx, dy = d_coordinates
	stage_x = (dx * self.A + dy * self.C) * self.scale_x
	stage_y = (dx * self.B + dy * self.D) * self.scale_y
	return (stage_x, stage_y)

def convert_to_d(self, s_coordinates):
	"""Convert stage coordinates into SEM coordinates"""
	stage_x, stage_y = s_coordinates
	stage_x /= self.scale_x
	stage_y /= self.scale_y
	dx = ((stage_y + stage_x * (-self.D/self.C))/
		  (self.B + self.A * (-self.D/self.C)))
	dy = ((stage_y - dx * self.B) / self.D)
	return (dx, dy)

def convert_to_v(self, d_coordinates):
	"""Convert SEM coordinates into viewport window coordinates. """
	dx, dy = d_coordinates
	return (int((dx - self.mv_dx_dy[0]) * self.mv_scale),
			int((dy - self.mv_dx_dy[1]) * self.mv_scale))
```
would become

```
def convert_to_s(self, d_coordinates):
	"""Convert SEM coordinates into stage coordinates"""
	dx, dy = d_coordinates
	stage_x, stage_y = utils.applyAffineT([dx], [dy], transform_sem_to_stage)
	return stage_x[0], stage_y[0]

def convert_to_d(self, s_coordinates):
	"""Convert stage coordinates into SEM coordinates"""
	stage_x, stage_y = s_coordinates
	dx, dy = utils.applyAffineT([stage_x], [stage_y], invertAffineT(transform_stage_to_sem))
	return dx[0], dy[0]

def convert_to_v(self, d_coordinates):
	"""Convert SEM coordinates into viewport window coordinates. """
	dx, dy = d_coordinates
	vx, vy = utils.applyAffineT([dx], [dy], transform_sem_to_viewport)
	return vx[0], vy[0]
```

btitze, 2019-08-13:
I have simplified the calculation I used in convert_to_d(). It was old code that worked well and that's why I never had a look at it again. I now use the standard rotation matrix convention, and multiply with the inverted matrix in convert_to_d(). It's essentially the same transformation, but now it can deal with rotation angles that are zero. I have renamed the variables and added comments, so it should be clear now what I'm doing.

templiert, 2019-08-13:
> it can deal with rotation angles that are zero

Yes, now working here with 
```
stage_scale_factor_x = 1
stage_scale_factor_y = 1
stage_rotation_angle_x = 0
stage_rotation_angle_y = 0
```
Thanks.

closed 2019-08-16 by btitze