SBEMimage/SBEMimage #15: magc: error in calculate_stage_move_duration after merging rotation features and with use_microtome=False

templiert, 2019-08-09:
simulation_mode = True
magc_mode = True
No problem with use_microtome = True, but the following error happening with use_microtome = False. 
```
Loading configuration file laptopOffline.ini ... Done.
Loading system settings file this_system.cfg ... Done.
Configuration loaded and checked: complete, no updates
Initializing SBEMimage. Please wait...
[........  ] 80%Traceback (most recent call last):
  File "sbemimage.py", line 143, in <module>
    main()
  File "sbemimage.py", line 134, in main
    VERSION)
  File "C:\Research\sbemimage\SBEMimage\src\main_controls.py", line 91, in __init__
    self.show_estimates()
  File "C:\Research\sbemimage\SBEMimage\src\main_controls.py", line 677, in show_estimates
    date_estimate) = self.stack.calculate_estimates()
  File "C:\Research\sbemimage\SBEMimage\src\stack_acquisition.py", line 301, in calculate_estimates
    max_offset_slice_number + max_interval_slice_number))
  File "C:\Research\sbemimage\SBEMimage\src\stack_acquisition.py", line 253, in calculate_for_slice_range
    x0, y0, x1, y1))
  File "C:\Research\sbemimage\SBEMimage\src\stage.py", line 96, in calculate_stage_move_duration
    return self._stage.calculate_stage_move_duration(from_x, from_y, to_x, to_y)
AttributeError: 'SEM_SmartSEM' object has no attribute 'calculate_stage_move_duration'
```

One thing that bugs me is that the error
- occurs in the head of magc-integration after merging the rotation features
- does not occur in magc-integration before merging the rotation features [(this commit)](https://github.com/SBEMimage/SBEMimage/commit/dd802d21dfd1f4b5a63463655680c4627a412338)
- does not occur in current master

I'm posting it in case you quickly see that it's from you Benjamin, if not, then it might be something on my side.

btitze, 2019-08-09:
It seems to be from a previous update, in which I improved the part that calculates estimates for a stack acquisition. But it's unclear to me why this error has not occurred earlier when you tested it with use_microtome = True.
In any case, I have now added the missing function in the SEM base class, so it should be available in SEM_SmartSEM, too. Can you try again with the code I have just pushed to the master?

templiert, 2019-08-09:
> It seems to be from a previous update, in which I improved the part that calculates estimates for a stack acquisition. But it's unclear to me why this error has not occurred earlier when you tested it with use_microtome = True.

looking at [the last time I merged from master](https://github.com/SBEMimage/SBEMimage/network) it must have happened after [this commit](https://github.com/SBEMimage/SBEMimage/commit/c250d25e15b93fe44453b6a884f943346c8136a1)

> In any case, I have now added the missing function in the SEM base class, so it should be available in SEM_SmartSEM, too. Can you try again with the code I have just pushed to the master?

perfect, it is working now, thanks Benjamin

closed 2019-08-09 by btitze