SBEMimage/SBEMimage #16: pivot is centered on tile (1,1) instead of on the center of the grid

templiert, 2019-08-09:
I set the center and rotation of grids with 

```
self.cs.set_grid_origin_s(idx, center)
self.gm.set_rotation(idx, angle)
```

but the pivot of the grid seems to be centered on the (1,1) tile of the grid instead of on the center of the grid:

[image]

which is confirmed when I add 180 degrees to the rotation :

[image]

btitze, 2019-08-12:
Every grid is anchored at tile (0, 0) - that is what I call the origin of the grid.
The pivot is also centred on tile (0, 0). I'm not sure why it would be (1, 1) in your case, but perhaps it is (0, 0)?
I chose (0, 0) for the pivot for stability and simplicity: There is always a tile (0, 0) in a grid and by choosing the origin as the pivot I don't have to keep a separate set of coordinates for the pivot. Also, when the user changes the size of the grid, the pivot would jump around if we define it to be the centre of the grid. Let's discuss this on the phone. We should also agree on a convention for the angle (clockwise, anti-clockwise?)

templiert, 2019-08-12:
> Every grid is anchored at tile (0, 0) - that is what I call the origin of the grid.

Ok.

> The pivot is also centred on tile (0, 0). I'm not sure why it would be (1, 1) in your case, but perhaps it is (0, 0)?

Sorry yes you're right, I meant (0,0)

> I chose (0, 0) for the pivot for stability and simplicity: There is always a tile (0, 0) in a grid and by choosing the origin as the pivot I don't have to keep a separate set of coordinates for the pivot.

I agree with having pivot = origin, but I believe that 

pivot = origin = center of grid

might be more practical (at least it was in my simple script) than

pivot = origin = center of tile (0,0).

> Also, when the user changes the size of the grid, the pivot would jump around if we define it to be the centre of the grid.

Yes, but the pivot does not jump around if pivot = origin = center of grid. May I mention how I did things to set rotated grids:

- [create a template grid](https://github.com/templiert/MagC/blob/master/EM_Imaging.py#L622-L651) centered on x=0,y=0 : I deal only with **centers** and store the centers of all tiles
- [rotate the grid (i.e., rotate the the centers of all tiles)](https://github.com/templiert/MagC/blob/master/EM_Imaging.py#L270-L273) around x=0,y=0
- translate the rotated grid to the section location (the translation vector is exactly the center of the section) 

> Let's discuss this on the phone.

ok yes, whenever you want

> We should also agree on a convention for the angle (clockwise, anti-clockwise?)

I'm fine with what you did.

templiert, 2019-08-15:
> I chose (0, 0) for the pivot for stability and simplicity

Ok I started from there then.

[The following functions](https://github.com/SBEMimage/SBEMimage/commit/87ba61ca8e3637dcee4a9b7f7da662effaa998fd)
```
gm.get_grid_center_s
gm.set_grid_center_s 
```
are now working:

[image]

One must remain careful about the following functions that change the center of the grid:
```
set_grid_size
set_rotation
set_overlap
set_row_shift
set_tile_size_selector
set_tile_size_px_py
set_pixel_size
```

btitze, 2019-08-16:
OK, thanks for the update!
When I tried out my rotation dialog box yesterday (with live preview) I noticed that indeed a rotation around tile 0 is not what one would intuitively expect, especially for larger grids.
So I'll allow two options for the pivot, and by default the rotation when using the rotation tool will be around the grid centre now. But internally, it still uses tile 0 as both origin and pivot. I'll push a number of updates later today or on the weekend!

closed 2019-09-18 by btitze