SBEMimage/SBEMimage #17: magc: high latency in viewport with hundreds of grids when grid label displayed

templiert, 2019-08-12:
When loading 430 grids ([these for example](https://github.com/SBEMimage/SBEMimage/blob/magc-integration/magc_433_sections.magc)), the latency of the viewport is low only if the labels are not displayed (even with 433 grids * 100 tiles/grid = 43 300 tiles). With labels displayed, the latency of zooming and panning is quite high.

Maybe the labels should be hidden during a user action (zooming, panning) and painted only at the end of the action ?

[image]

btitze, 2019-08-13:
Thanks for pointing this out! Yes, it appears that drawing the tile and grid labels takes a lot of time.

I tried to load your sections and noticed that the grid numbers are useless clutter when zoomed out.
I have pushed a few changes to deal with this: When the number of grids is larger than 10, the labels will only show when sufficiently zoomed in (you can adjust the threshold of mv_scale if you want). 
I also implemented the change you suggested: During dragging or panning actions, label drawing is now suppressed.
Another thing I noticed (at least on my computer) is that the grids disappear when zoomed out completely because the tile rectangles become to small to be drawn. The updated code ensures that in this case, each grid will be drawn as a single pixel, so that all grids will always be visible (at least as a point cloud).

templiert, 2019-08-13:
> I tried to load your sections and noticed that the grid numbers are useless clutter when zoomed out.

I agree.

> When the number of grids is larger than 10, the labels will only show when sufficiently zoomed in (you can adjust the threshold of mv_scale if you want).

It works, thanks.

> I also implemented the change you suggested: During dragging or panning actions, label drawing is now suppressed.

It works for dragging, thanks, but not for zooming. A suggestion:
- after a zoom, paint everything first before painting the labels
- if we zoom in/out several times in a row with the mouse wheel, it is hanging. Maybe paint the labels only 0.5s after a zoom and cancel the repainting if there is again a zoom before repainting is done ?

> During dragging or panning actions, label drawing is now suppressed.

Nice, it's working.

> The updated code ensures that in this case, each grid will be drawn as a single pixel, so that all grids will always be visible (at least as a point cloud).

Excellent, I did not think of it.

templiert, 2019-08-15:
Also, selected sections in the MagC list are highlighted in red in the viewport: this process is slow when labels are displayed. As you mentioned @btitze, there could be a minimal painting function that updates only some grid parameters like color.

folterj, 2024-08-27:
This seems to be tailored now to give good responsiveness.

closed 2024-08-27 by folterj