SBEMimage/SBEMimage #18: Windows 10 display scaling

btitze, 2019-08-13:
When running SBEMimage under Windows 10 with display scaling set to 100%, the PyQt GUI looks a bit different, but the proportions are correct and all elements work as expected.

When setting the scaling to a different value (e.g. 150% or 200%), the fonts don't scale correctly -- many strings in the GUI are cut off.

templiert, 2019-08-13:
Add

```
import PyQt5
PyQt5.QtWidgets.QApplication.setAttribute(PyQt5.QtCore.Qt.AA_EnableHighDpiScaling, True)
```
in sbemimage.py as [advised here](https://www.reddit.com/r/learnpython/comments/69vm4t/pyqt5_and_high_resolution_monitors/) ?

Some [more info here](https://doc.qt.io/qt-5/highdpi.html) ...

btitze, 2019-08-14:
Thanks, I'll look into that.

closed 2019-08-16 by btitze