SBEMimage/SBEMimage #19: this_system.cfg is overriding .ini file ?

templiert, 2019-08-13:
I change these parameters in the .ini file:

```
stage_scale_factor_x = 1
stage_scale_factor_y = 1
stage_rotation_angle_x = 0
stage_rotation_angle_y = 0
```
They get correctly loaded in sbemimage. Then I close sbemimage and choose to update the .ini file in the closing dialog: this reverts the .ini file to the values stored in this_system.cfg:

```
sem_calibration_data = {"1500": [1.12612, 1.01419, 0.00225, 0.00507]}
```

```
stage_scale_factor_x = 1.12612
stage_scale_factor_y = 1.01419
stage_rotation_angle_x = 0.00225
stage_rotation_angle_y = 0.00507
```
Is it expected that this_system.cfg overrides the .ini file or should the .ini file be updated ?

btitze, 2019-08-14:
Yes, it's designed that way. The system configuration always overrides the user/project configuration in the ini files. Even when you changed these settings in the .ini file and loaded it in SBEMimage, it actually used the settings from the system config file.

The system configuration settings are device-specific and should be shared by all users that work on the same microscope/3View. So, you should change these settings directly in the system configuration file. When using a microtome, you can also change them through the GUI, but this is not implemented yet for SEM stages.

templiert, 2019-08-14:
> Yes, it's designed that way. The system configuration always overrides the user/project configuration in the ini files.

> The system configuration settings are device-specific and should be shared by all users that work on the same microscope/3View. So, you should change these settings directly in the system [...]

Ok, makes sense, but see below.

> Even when you changed these settings in the .ini file and loaded it in SBEMimage, it actually used the settings from the system config file.

No, the .ini file gets loaded ("They get correctly loaded in sbemimage."). That is why I was surprised that .ini was finally overriden by .cfg while we have the opposite at startup (for these parameters at least).

.ini file:
```
stage_scale_factor_x = 1
stage_scale_factor_y = 1
stage_rotation_angle_x = 0.4
stage_rotation_angle_y = 0.4
```
this_system.cfg: 
```
sem_calibration_data = {"1500": [1, 1, 0, 0]}
```
gives the following:

[image]

btitze, 2019-08-15:
You are right -- it does apply the correct settings from the system configuration to the 3View stage, but not to the SEM (Merlin) stage. I've fixed it now.

closed 2019-08-15 by btitze