SBEMimage/SBEMimage #23: magc: viewport horizontally flipped compared to smartsem

templiert, 2019-09-11:
After wafer calibration, the coordinate systems are correct (right click in viewport on a section visible in the wafer image moves the stage correctly to that section). 

A cosmetic issue remains: the viewport is horizontally flipped compared to the smartsem view

[image]

[image]

btitze, 2019-09-26:
Another issue with the rotation angle (which may be related):
I've noticed that you are using 
self.sem.set_scan_rotation(360 - theta)
But I'm using 
self.sem.set_scan_rotation(theta)

and it seems to work on our Merlin.

templiert, 2019-09-26:
> I've noticed that you are using
> self.sem.set_scan_rotation(360 - theta)

No, I'm using self.sem.set_scan_rotation(theta) like you [see here](https://github.com/SBEMimage/SBEMimage/blob/magc-integration/src/stack_acquisition.py#L1697)

(I believe I had originally set (360-theta) before the implementation of the rotations because I had been anticipating an issue that finally did not occur)

templiert, 2019-09-26:
editing is currently broken, I skipped a bracket:

No, I'm using self.sem.set_scan_rotation(theta) like you [(see here)](https://github.com/SBEMimage/SBEMimage/blob/magc-integration/src/stack_acquisition.py#L1697)

btitze, 2019-09-26:
Oops, I meant the opposite. The lines got swapped:

I am actually using `self.sem.set_scan_rotation(360 - theta)`

We should look into that. I don't understand why it should be different unless it's a machine-specific issue.