SBEMimage/SBEMimage #26: added "Open grid settings of grid xxx" in viewport right click menu

templiert, 2019-09-13:
For the record: added [in this commit](https://github.com/SBEMimage/SBEMimage/commit/3c41587f2da62b58ce990044b55c0cae38efafd8). If magc later merged to master, it will affect standard sbemimage.
I can add a `if magc_mode == 'True'` if not useful.

[image]

templiert, 2019-09-13:
sorry, with a small correction in this [commit too](https://github.com/SBEMimage/SBEMimage/commit/145f00dfba8e4bb4694050f7dd94675f481f8358#diff-12d73a6784b8aa6ebf875009368e0d76L966) (setting the current_grid to the grid_number that has been passed as argument so that the buttons 'save setting for grid xxx' and 'delete grid xxx' also update the displayed grid number)

btitze, 2019-09-13:
Looks good! I think this could be a standard option in the menu.

btitze, 2019-11-11:
Now included after the recent merge.

closed 2019-11-11 by btitze