SBEMimage/SBEMimage #28: magc: acquisition of large tiles 16k x 12k triggers 'Load image error'

templiert, 2019-09-17:
Acquiring several grids in a row is working with small tiles but when switching to a larger tile size, the very first tile is failing repetitively: 

```
2019-09-17 21:09:44.97 | CTRL  : SBEMimage Version 2.0 (R2019-07-09)
2019-09-17 21:09:52.56 |       433 MagC sections have been loaded.
2019-09-17 21:10:19.12 |       433 MagC sections have been loaded.
2019-09-17 21:12:07.98 | CTRL  : RESET command received.
2019-09-17 21:12:17.92 | CTRL  : Stack started.
2019-09-17 21:12:17.92 | CTRL  : Settings saved to disk.
2019-09-17 21:12:19.16 | SEM   : WD/STIG_XY: 8.284338, -4.164715, -2.532160
2019-09-17 21:12:19.16 | CTRL  : ****************************************
2019-09-17 21:12:19.16 | CTRL  : slice 0, Z:41803.218
2019-09-17 21:12:19.16 | CTRL  : Grid 0, number of active tiles: 1
2019-09-17 21:12:19.16 | CTRL  : Starting acquisition of active tiles in grid 0
2019-09-17 21:12:20.84 | SEM   : Locked magnification: 1751.0
2019-09-17 21:12:21.48 | SEM   : WD/STIG_XY: 8.284338, -4.164715, -2.532160
2019-09-17 21:12:21.48 | 3VIEW : Moving stage to position of tile 0.0
2019-09-17 21:12:24.52 | CTRL  : Running SmartSEM AF procedure (focus+stig) for tile 0.0
2019-09-17 21:12:37.58 | CTRL  : Completed SmartSEM autofocus + autostig procedure.
2019-09-17 21:12:38.84 | SEM   : Acquiring tile at X:62570.481, Y:66154.679
2019-09-17 21:15:53.86 | CTRL  : Trying again to image tile.
2019-09-17 21:15:53.86 | 3VIEW : Moving stage to position of tile 0.0
2019-09-17 21:15:56.90 | CTRL  : Running SmartSEM AF procedure (focus+stig) for tile 0.0
2019-09-17 21:16:15.82 | CTRL  : Completed SmartSEM autofocus + autostig procedure.
2019-09-17 21:16:17.07 | SEM   : Acquiring tile at X:62570.481, Y:66154.679
2019-09-17 21:19:32.15 | CTRL  : Trying again to image tile.
2019-09-17 21:19:32.15 | 3VIEW : Moving stage to position of tile 0.0
2019-09-17 21:19:35.19 | CTRL  : Running SmartSEM AF procedure (focus+stig) for tile 0.0
2019-09-17 21:19:49.15 | CTRL  : Completed SmartSEM autofocus + autostig procedure.
2019-09-17 21:19:50.41 | SEM   : Acquiring tile at X:62570.481, Y:66154.679
2019-09-17 21:23:06.62 | CTRL  : Settings saved to disk.
2019-09-17 21:23:06.88 | CTRL  : Load image error
2019-09-17 21:23:06.90 | CTRL  : Stack paused.
```

Is it not a timing issue because it takes more time to save the image (0.25 GB) and a os.path.isfile check is happening too early ? Maybe [this one in stack_acquisition](https://github.com/SBEMimage/SBEMimage/blob/master/src/stack_acquisition.py#L1566) ?

I saw that 
- the image file is appearing for half a second in the folder
- then `CTRL  : Trying again to image tile.` is written
- then the file disappears

The saving folder is on a local hard drive (d\temp)

btitze, 2019-09-18:
My guess is that the cycle time is too short. You can set this time in your system configuration file. (Make sure SBEMimage is closed when editing this file.)
These are the default times:

`cycle_time = {"0": [0.085, 0.13, 0.21, 0.38, 0.71, 1.4, 2.7, 5.3, 10.3], "1": [0.044, 0.054, 0.075, 0.12, 0.20, 0.37, 0.70, 1.4, 2.6], "2": [0.23, 0.40, 0.73, 1.4, 2.7, 5.4, 10.5, 20.6, 40.7], "3": [0.46, 0.83, 1.6, 3.1, 6.1, 12.0, 23.3, 46.0, 90], "4": [0.77, 1.4, 2.8, 5.4, 10.8, 21.0, 41.1, 84, 162], "5": [0, 0, 0, 0, 0, 0, 0, 0, 0], "6": [1.6, 3.1, 6.1, 12.2, 23.9, 46.6, 90, 180, 366], "7": [4.4, 7.1, 12.4, 23.1, 43.5, 84, 162, 324, 648], "8": [8.6, 14.6, 26.6, 50.2, 96, 186, 366, 732, 1452], "9": [14.1, 24.8, 46.1, 90, 168, 330, 648, 1296, 2586], "10": [29.1, 53.2, 102, 192, 372, 732, 1458, 2910, 5760], "11": [49.5, 90, 174, 336, 660, 1302, 2592, 5040, 10440]}
`

The keys correspond to the frame size selectors and the cycle times in the lists to the different dwell times (from 0.05 microseconds in increasing order). I have noticed that these cycle times may vary between systems.
Check the entry for your target resolution and dwell time and measure it, and increase it if necessary. Note that some cycle times displayed in SmartSEM are wrong (this bug has already been reported).

templiert, 2019-09-24:
> My guess is that the cycle time is too short.

It does not seem to be the problem.

I am using tile_selector = 9, (16384, 12288), dwell time 800 ns, cycle time 2.8 min = 168 s.

Manually measuring the cycle time gives approximately 168 s, which corresponds to the config.

I increased the config to 190 s to be sure, the following is happening:

- after ~170 s the scanning of the frame is done 
- 20 s later (after 190 s), the grab command is sent (the cycle waiting is done in [`acquire_frame`](https://github.com/SBEMimage/SBEMimage/blob/master/src/sem_control.py#L538))
- ~20 s later (after 210 s), the image file briefly appears for less than a second in the windows file browser, a snapshot of the image can be seen which looks ok, then the file immediately disappears and `CTRL  : Trying again to image tile` is displayed

It looks like the file gets actively deleted ...

templiert, 2019-09-24:
I'm printing the output of the [img_inspector.process_tile here](https://github.com/SBEMimage/SBEMimage/blob/master/src/stack_acquisition.py#L1567-L1575)

which gives

```
range_test_passed, slice_by_slice_test_passed, tile_selected, load_error, grab_incomplete, frozen_frame_error 
False False False True False False
```

- it looks like something wrong is happening in the img_inspector
- all tile resolutions below (16384, 12288) are ok but when the tile has this dimension or above it is failing
- it is failing immediately at the first tile of the first grid

templiert, 2019-09-24:
The tile is actually in the temp\tiles\rejected folder

Ok I found the problem: the image tile is too big for PIL.Image 

```
DecompressionBombError('Image size (201326592 pixels) exceeds limit of 178956970 pixels, could be decompression bomb DOS attack.')

```
Maybe a display of the exception in the try/except (even if not catching specifically) instead of the silent catch would have helped:

```
        try:
            img = Image.open(filename)
        except Exception as e:
            print(repr(e))
            load_error = True
```
is safer than
```
        try:
            img = Image.open(filename)
        except:
            load_error = True
```

[This post](https://stackoverflow.com/questions/51152059/pillow-in-python-wont-let-me-open-image-exceeds-limit#comment92326274_51152514) solves the issue by adding 

```
PIL.Image.MAX_IMAGE_PIXELS = None
```

btitze, 2019-09-25:
Many thanks for tracking this down and fixing it!
(We had never used tiles of that size because the aberrations at the edges were too strong.)

closed 2019-09-25 by btitze