SBEMimage/SBEMimage #29: frequently check for the end of frame acquisition instead of using cycle_time ?

templiert, 2019-09-24:
Instead of relying on potentially wrong `cycle_times` to guess when the acquisition of a frame is finished, would it not be simpler and possibly more reliable to check every e.g. 0.5 second whether the acquisition is done ? These regular checks seem innocuous.

At first I did not understand why you were using these cycle_time as I could not remember using them.

I used to do it like this ([here](https://github.com/templiert/MagC/blob/master/EM_Imaging.py#L512-L513)):
```
    freezeAtEndOfFrame()
    logger.info('Scanning ...')
    while a.Get('DP_FROZEN')[1] == 'Live':
        time.sleep(sleepScanning)
        
    logger.info('Tile ' + str(sectionIndex) + '-' + str(tileIndex) + ' acquired. Now grabbing ...')
    tilePath = os.path.join(folder, 'Tile_' + str(tileIndex[0]) + '-' +  str(tileIndex[1]) + '.tif')
    a.Grab(0, 0, 1024, 768, 0, tilePath)
```

templiert, 2019-09-24:
Although actually using your cycle_times is useful to estimate the length of acquisitions.

Maybe a mix of the two approaches would be good. Make estimates with cycle_times but use the loop above to prevent issues coming from slightly inaccurate cycle_times.

btitze, 2019-09-25:
There were several reasons for using the cycle times when I started implementing this a couple of years ago -- I can explain the next time we talk on the phone. 

Performing a DP_FROZEN check (repeatedly in short intervals) after the cycle_time is a good idea. The program could display a warning if the cycle_time is too short.

templiert, 2019-09-25:
> There were several reasons for using the cycle times when I started implementing this a couple of years ago

Sure, no doubt about it

> Performing a DP_FROZEN check (repeatedly in short intervals) after the cycle_time is a good idea. The program could display a warning if the cycle_time is too short.

That sounds good. The program could even self-calibrate the cycle_time and suggest a better value "Cycle time xxx was too short. Do you want to update the cycle time to yyy ?"

btitze, 2019-11-11:
I have implemented this now in a basic way. A self-calibration feature could be added at a later stage.

closed 2019-11-11 by btitze