SBEMimage/SBEMimage #37: Setting autofocus positions of a grid (mainly for MagC)

templiert, 2020-04-22:
What is the current way to set autofocus locations for a grid? 

For example for the configuration below, with autofocus points located at the * and the imaging part shown with dashes, we would need to 
- define a grid much larger (e.g. 5x5) than the original imaging grid (e.g. 2x2)
- set only the central 2x2 tiles as "active" for imaging
- select tiles that will be used for autofocus. Here it also makes a constraint on the possible autofocus locations: they have to be at the center of some tiles in the grid, correct?
*               *
   -------
  |           |
   -------
*                *

I am tempted to implement custom autofocus locations for each grid and to display them with a simple point in the viewport. Any reason not to do that?

btitze, 2020-04-22:
At the moment, it's tile-based. Every tile in a grid can be active as an autofocus reference tile or not. If a tile is active, the autofocus will be run for that tile, by default in the centre of the tile.

So, you're right, for acquiring a 2x2 grid and focusing outside the active region, you would have to proceed as you describe it, by enlarging the grid and selecting surrounding tiles for the autofocus.

Your suggested approach sounds interesting and would provide extra flexibility.
You could have an extra variable in class Grid, for example self.autofocus_locations, that holds a list of locations (absolute stage coordinates or coordinates relative to the grid?). Or a similar approach.
 
But, let's say you save four locations per grid and then run the autofocus at these locations before acquiring the grid. How would you then set the focus for acquiring the tiles? Would you calculate the average parameters from these four locations? Or set each tile to the parameters of the closest focusing location?

templiert, 2020-04-22:
Thanks for weighing in.

> You could have an extra variable in class Grid, for example self.autofocus_locations, that holds a list of locations

Agreed. I am wondering whether this Grid.autofocus_locations could be the central holder for both the current system and the custom location system. Maybe it would be easier to use Grid.autofocus_locations separately only for the special MagC case. I do not have a good enough overview of all the autofocus approaches currently implemented.

> absolute stage coordinates or coordinates relative to the grid?

I would tend to choose relative to the (non-rotated) grid.

> Would you calculate the average parameters from these four locations?

With more than 3 focus points I would calculate an interpolation for each grid (simple plane or b-splines). I had implemented a simple plane [in my old imaging script](https://github.com/templiert/MagC/blob/master/EM_Imaging.py#L329-L353).

> Or set each tile to the parameters of the closest focusing location?

With less than 3 points yes probably.

btitze, 2020-04-22:
The current tile-based autofocus system uses a central list of tiles in the autofocus section of the configuration file, which is read and processed by grid_manager. It works well and is deeply integrated into the Viewport display and other parts of SBEMimage.

I suggest that you go ahead and implement the functionality you described as a feature for MagC, so that autofocus_locations would be only for these custom locations for now. We could then discuss at a later stage if it makes sense to fuse the two approaches, or keep them separate but complementary.

btitze, 2020-04-22:
I also remember this nice suggestion you made a while ago:
https://github.com/SBEMimage/SBEMimage/issues/21

Perhaps this could be added as a general routine for both custom and tile locations.