SBEMimage/SBEMimage #43: Implementation on Hitachi SEM

robbmcleod, 2020-07-06:
Hello I work with Hitachi High-tech and we had an inquiry from a customer on what would be required to implement SBEMimage on their system. It appears to me at first glance that you have the abstract base class `src.sem_control.SEM` that would have to be implemented, and then the `cfg/system.cfg` would have to be customized? Are there any other issues I would have to account for?

Thanks in advance for any assistance.

btitze, 2020-07-07:
Hi, thanks for getting in touch!
Yes, exactly as you describe it. But we will probably have separate files for hardware-specific implementations soon. 
Currently sem_control.py contains both the SEM base class and the SmartSEM class for ZEISS microscopes. A class to control the FEI Quanta is under development.

In your case, you would need to implement the class 'SEM_Hitachi' (or whatever name would be appropriate) and add custom parameters to system.cfg. If possible, existing parameters in system.cfg should be used as 'generic' parameters for all microscope models.

Perhaps we could discuss further details on the phone? (You can send an e-mail to benjamin.titze AT fmi.ch)

robbmcleod, 2020-07-08:
Will do, thanks.

closed 2020-07-08 by robbmcleod