SBEMimage/SBEMimage #46: Offline Installation doc?

cwood1967, 2020-08-18:
Your new readthedocs page is great, thanks.

We might be a special case, but our EM uses a computer with Windows 7 that is not allowed to access the internet. This presents problems when trying to pip install or create conda environments. There are ways around this, but it takes some time to figure everything out the first time doing it. 

Do think this issue is common enough to have a special section in your documentation?
I am happy to help out if you thing so.

~Chris

btitze, 2020-08-19:
Hi Chris,

yes, I'd be interested in how you've done this in your case. Can you send me an e-mail or post the info here?
There may be other labs with the same problem (no network access on setup computers). 

A while ago, I tried to freeze the current code and dependencies into an executable, but that has not worked and I didn't have time to troubleshoot it. I was also looking into using an installer that would automatically create an environment, install the dependencies, and copy the source code and other files, to make it easier for users to get started. 

Best,

Benjamin

cwood1967, 2020-10-05:
Hi Benjamin,

Below are some details about how I did the offline installation. Let me know if you have questions.
Chris 

Offline Installation
====================

Install the latest version of Anaconda or Miniconda (python 3.6 or greater) 
on the the offline machine.

On a computer with internet access, create a conda environment that matches the
conda environment on the acquisition computer: the architecture, operating system,
and the python version. Install pip using conda:

```bash
conda install pip
```
Use the pip download option to download the required python wheel files rather than installing
them (I have tried using the requriements.txt file to do this in one
command,but haven't figured it out). Make a new directory, cd into it, and do the pip downloads:

```bash
mdkir pip_downloads
cd pip_downloads
pip download PyQt5
pip download colorama
pip download validators
pip download validate_email
....
```

Copy this folder and the files to the offline computer over the local network or
with a USB drive.

Then on the offline computer, for all whl or tgz files in the folder:

```bash
pip install <filename1>
pip install <filename2>
```

### Conda Pack
Alternatively, you can also use conda-pack to copy an entire environment from one computer to another:
https://conda.github.io/conda-pack/

btitze, 2020-10-08:
Thanks for sharing!

closed 2020-10-25 by btitze