SBEMimage/SBEMimage #67: Baked-in Parameters for SEM classes

robbmcleod, 2021-05-04:
I'll try and collect here where I find fixed values that are incompatible with implementations other than the Zeiss class.

1. There are a number of locations in the core modules where the filename extension is implicitly assumed to be `.tif`.

As a temporary work-around, inside `SEM_SU7000.acquire_frame` I load the BMP with `imageio` and then save it back as TIFF.

2. In `main_controls_dlg_windows.py::calibration_images_acq_thread` the `frame_size_selector` defaults to 3 or 4 and cannot be set via the `.ini` file.

robbmcleod, 2021-05-05:
`main_controls_dlg_windows.py::StageCalibrationDlg.__init__` sets `self.comboBox_dwellTime` to index of `4`. I changed this from,

    self.comboBox_dwellTime.setCurrentIndex(4)

to,

    dwell_index = self.sem.DWELL_TIME.index(self.sem.grab_dwell_time)
    self.comboBox_dwellTime.setCurrentIndex(dwell_index)

btitze, 2021-05-22:
Thanks for pointing this out! I'll add defaults for both the store resolution (frame size) and the dwell time to the system configuration and the device presets. Let's discuss the image file name extensions separately - I'll get in touch next week about this and related issues.

robbmcleod, 2021-06-01:
Ok, pulled the latest changes, have found another one when trying the 'Image Stub' feature.

```py
Exception hook caught: Traceback (most recent call last):
  File "C:\Temp\SBEMimage\src\viewport.py", line 2747, in _vp_open_stub_overview_dlg
    dialog = StubOVDlg(centre_sx_sy,
  File "C:\Temp\SBEMimage\src\viewport_dlg_windows.py", line 70, in __init__
    self.update_dimension_and_duration_display()
  File "C:\Temp\SBEMimage\src\viewport_dlg_windows.py", line 135, in update_dimension_and_duration_display
    cycle_time = self.ovm['stub'].tile_cycle_time()
  File "C:\Temp\SBEMimage\src\grid_manager.py", line 898, in tile_cycle_time
    return self.sem.CYCLE_TIME[size_selector][scan_speed] + 0.2
IndexError: list index out of range
```

folterj, 2024-08-27:
Assuming this was resolved, closing.

closed 2024-08-27 by folterj