SBEMimage/SBEMimage #68: SEM selection in `system.cfg` overrides the .ini file

robbmcleod, 2021-06-21:
The device selection in the .ini file does not appear to do anything, only the selection made in the `system.cfg` file 

E.g. if in `system.cfg`:
```ini
[device]
sem = UNKNOWN
microtome = UNKNOWN
sem_recognized = ["ZEISS Merlin", "ZEISS Sigma", "ZEISS GeminiSEM", "ZEISS Ultra Plus", "ZEISS MultiSEM", "TESCAN MIRA", "TESCAN CLARA", "TESCAN MAGNA", "Hitachi SU7000", "Mock SEM"]
microtome_recognized = ["Gatan 3View", "ConnectomX katana", "GCIB", "Mock Microtome"]
katana_com_port = COM1
sem_ip_address = 
sem_port = 
```

whereas my .ini file has:

```ini
[sem]
device = Hitachi SU7000

[microtome]
device = ConnectomX katana
```
Then it always loads UNKNOWN, None for the SEM, Microtome combo.

robbmcleod, 2021-06-21:
Ok having investigated further I think your expectation is that `system.cfg` should be written to. Personally I would think this file should be immutable. Users have to modify this file to operate the machine, and this file is included in Git source control, and that's a conundrum. 

My suggestion would be to `git mv cfg/system.cfg src/default.cfg`, and likewise for `cfg/device_presets.cfg` and `cfg/default.ini`. Remove all files in `cfg` from source control. Then on start-up if there is no `cfg/system.cfg`, a dialog should prompt the user to select new SEM/Microtome presets and then it writes a `cfg/system.cfg` as appropriate.

btitze, 2021-06-21:
Thanks for bringing this up! I agree that the current way the configuration is loaded may be confusing...

Let me clarify: `system.cfg` and `default.ini` should actually never be written to by the user. They are the only configuration files that are tracked because they are modified/updated by developers and contain documentation of the entries as comments. All other configuration files in `cfg` (those created by the user) are not tracked. 
`system.cfg` and `default.ini` are the templates for the initial configuration and also used for checking the validity of configuration files and ensuring that older configuration files can be loaded.
I will consider moving these templates and the device_presets to `src` or some other folder to make their special role clearer and prevent users from accidentally modifying them. This way, `cfg` would only contain user-generated files, that would indeed be cleaner.

At the moment, on a setup with a new installation, the user can start SBEMimage and select default.ini plus the SEM/microtome device presets. The user can then save a new configuration file (.ini) and will be prompted to also save a new system configuration file (.cfg). New .ini configuration files can be created by opening existing ones and saving them under a new name.

robbmcleod, 2021-06-21:
Ok, I think I misunderstood what you said in the call about the new `device_presets.cfg`, I thought that precluded the need for an changes to `system.cfg` or my copy of `su7000.cfg`.

closed 2021-06-22 by robbmcleod