SBEMimage/SBEMimage #69: Imagelist for overviews

K-Meech, 2021-07-19:
It would be useful to be able to easily link an overview image with its corresponding config file (from the logs directory).
This is already possible for tiles via the 'imagelist' files, that carry the same id as the config file.

@btitze Is there a way to find the corresponding config file for an overview? If not, could the equivalent of 'imagelist' be made for overviews? (I'm happy to implement this, if it would be useful)

btitze, 2021-07-20:
Yes, that would be useful!
For each run -- when the user clicks "Start" until the next "Pause/stop" -- a timestamp is attached to the current config file, the image_list file, the log file, and a few additional files.
This same timestamp could be used for an "imagelist_ov" file that lists all the OV images that have been acquired during a run.

Sure, if you'd like to implement this, please go ahead! :) There is a function called register_accepted_tile() in acquisition.py that is called when a tile is accepted. You could follow the same approach for overviews with a new function such as register_accepted_ov() that writes the filenames and positions into a new imagelist_ov file. I would probably keep two separate imagelist files for tiles and overviews. If you'd like to creat

K-Meech, 2021-07-21:
Ok - great! I'll put a draft together in my fork of the repo and make a PR against the 'dev' branch :)

closed 2021-08-23 by K-Meech