SBEMimage/SBEMimage #71: Target depth option in acquisition settings

K-Meech, 2021-08-23:
It would be useful to be able to give a target depth (in micrometer), rather than only the target number of slices. This is useful if e.g. you want to cut thick for a certain number of slices, then thinner, all aiming at the same final depth.

Perhaps there could be a dropdown in the Acquisition settings window to choose between a target number of slices vs a target depth?

@btitze Any objections / suggestions to this?
I'll start looking into it / making a rough PR.

btitze, 2021-08-23:
I agree that such an option would be useful! It would be nice to disable (grey out) the field that is not being used when the choice is updated by the user (target number vs target depth). You can also capture the settings in the session configuration by introducing two parameters such as target_depth and a bool such as use_target_depth in the config file.

One thing to note here: For the ∆Z "depth already cut" (total_z_diff is session configuration), I've only taken into account cuts made during the acquisition cycle. So if the user pauses, and then uses the Approach Dialog to cut further down, this is currently not included in the total.

K-Meech, 2021-08-24:
Ok - great! There's a working draft on the 'embl-dev' branch now. I'll try it out on the microscope (probably next week), then make a PR

closed 2021-09-07 by K-Meech