SBEMimage/SBEMimage #74: shapely errors with latest version

K-Meech, 2021-09-23:
Installing SBEMimage dependencies via ```pip install -r requirements.txt``` (from dev branch), leads to errors on running sbemimage.py. Error below:

```
Traceback (most recent call last):
  File "C:/Users/meechan/Documents/Repos/SBEMimage/src/sbemimage.py", line 47, in <module>
    from main_controls_dlg_windows import ConfigDlg
  File "C:\Users\meechan\Documents\Repos\SBEMimage\src\main_controls_dlg_windows.py", line 47, in <module>
    import utils
  File "C:\Users\meechan\Documents\Repos\SBEMimage\src\utils.py", line 28, in <module>
    from shapely.geometry import Polygon
  File "C:\Users\meechan\Anaconda3\envs\base-sbemimage\lib\site-packages\shapely\geometry\__init__.py", line 4, in <module>
    from .base import CAP_STYLE, JOIN_STYLE
  File "C:\Users\meechan\Anaconda3\envs\base-sbemimage\lib\site-packages\shapely\geometry\base.py", line 19, in <module>
    from shapely.coords import CoordinateSequence
  File "C:\Users\meechan\Anaconda3\envs\base-sbemimage\lib\site-packages\shapely\coords.py", line 8, in <module>
    from shapely.geos import lgeos
  File "C:\Users\meechan\Anaconda3\envs\base-sbemimage\lib\site-packages\shapely\geos.py", line 154, in <module>
    _lgeos = CDLL(os.path.join(sys.prefix, 'Library', 'bin', 'geos_c.dll'))
  File "C:\Users\meechan\Anaconda3\envs\base-sbemimage\lib\ctypes\__init__.py", line 364, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: [WinError 126] The specified module could not be found
```

This used to work fine, so perhaps there is some compatibility issue with the latest version? (Installing shapely from conda instead ```conda install shapely``` works fine)

folterj, 2021-09-23:
Hi!
So the C counterparts (geos C library) should get installed fine using either pip or conda
I found this: https://stackoverflow.com/questions/13144158/python-geos-and-shapely-on-windows-64
In particular - 'Make sure you install the version that matches your Python installation, NOT your OS. I made the mistake of installing the 64 bit shapely since I'm running 64 bit Windows, but my Python is 32 bit, and it gave me a similar "geos_c.dll" is not found'
If you're able to reproduce the error that's a good start at least!

K-Meech, 2021-09-23:
Ok - interesting! I also found this: https://github.com/Toblerity/Shapely/blob/master/FAQ.rst#i-installed-shapely-in-a-conda-environment-using-pip-why-doesnt-it-work which explains why it doesn't work for me (I'm using conda environments at the moment). Also there's this here: https://github.com/Toblerity/Shapely/issues/1105#issuecomment-802856196, so it seems all these issues should get fixed in version 2.0!

closed 2024-08-27 by folterj