SBEMimage/SBEMimage #83: style: use of built-ins {any,all} for readability

templiert, 2021-12-11:
Creating an issue because this style commit b7510a486dc466fc8897a6b8583986d937f50576 touches the core acquisition functions. I was digging into the grid acquisition flow and I felt that many checks like 
```
    if (self.use_autofocus and self.autofocus.method == 1
        and self.gm[grid_index][tile_index].autofocus_active):
``` 
would be more readable using all():
```
    if all([
        self.use_autofocus,
        self.autofocus.method==1,
        self.gm[grid_index][tile_index].autofocus_active,
        ]):
```
I only edited 
```
acquire_all_grids,
acquire_grid,
acquire_tile
```
Any comment?

btitze, 2021-12-14:
No objections - I agree it's a bit more readable.

templiert, 2021-12-16:
Ha, there's a catch. 

all() and any() short-circuit the evaluation of iterables (e.g. all() stops evaluating as soon as it finds a False). But the list here gets evaluated entirely first. Reverted two checks here ab5fe35d76453c5cecf66892e4e226cfcd830e62.

```
        if all([
            self.pause_state != 1,
            self.acq_interrupted,
            self.acq_interrupted_at[0] in self.grids_acquired,
            ]):
```
back to 

```
        if (self.pause_state != 1
            and self.acq_interrupted
            and self.acq_interrupted_at[0] in self.grids_acquired):
```

closed 2022-01-23 by btitze