SBEMimage/SBEMimage #97: Set specific WD/Stigmator globally across tracked tiles

TomasGancarcik, 2022-07-01:
Feature should allow to set specific working distance, stigX and stigY values across all tracked tiles within specific grid. This will dramatically increase the user experience when new, identical focus or stigmator setting is required for many (tracked) tiles. 

Design proposal: New buttons/setters for setting global WD, global stigX and stigY within 'Focus tool' -> 'Set' dialogue window. The values shall be taken from current WD/Stig values typed within this dialogue window. Confirmation message should appear after pressing each of this buttons/setters. Feature should work on-the-fly (no pausing of the acquisition required). The values should be updated with immediate action, which means all subsequent tiles within actual grid acquisition will be acquired with new WD and/or StigX and/or StigY.

folterj, 2024-11-21:
Priority increase

folterj, 2025-02-18:
I understand a possible solution could be to have a secondary dialog from the grid settings, but this would imply adding additional value input fields.
A simpler solution would be using the autofocus tool (tab) which has all the functionality and input fields, and allow the tile selector to be unset, or add an option 'all tiles' instead of 'select a tile', which would then apply setting to all tiles.
[image]

ganctom, 2025-02-19:
I have already implemented a solution to that. Will try to make pull request.

closed 2025-02-19 by ganctom

folterj, 2025-02-24:
Thank you @ganctom, indeed it would be great if you are able to make a PR on the dev branch!
We realise there have been many changes on the dev branch including a recent refactoring of the code structure, so hopefully this works. Alternatively if this is too problematic, we could consider reproducing the specific changes on the current code. Please let us know if we can help.